deepnest-io/Deepnest #3: SVG->DXF Conversion

bosd, 2023-04-16:
As discussed in:  https://github.com/Jack000/Deepnest/issues/7 and https://github.com/Jack000/Deepnest/issues/107
Deepnest calls to convert.deepnest.io for the conversion of svg to dxf.

A commonly reqeusted feature is offline conversion.

This commit on the fork of @awaistkd states support for offline dxf to svg conversion.
Can someone with more programming skills look into this??

https://github.com/Jack000/Deepnest/commit/3ce7ce18eb288d6f6c78450e087bbc615b2c01af

cmidgley, 2023-04-17:
I've looked briefly at this.  The ability to add an interface to export to different formats is straightforward, but the challenge is in the complexity of DXF itself.  There are a lot of open-source tools out there that work in this area so perhaps somebody could start by finding the least-effort library to convert SVG to DXF and try it by hand on Deepnest-generated files to see how well it works (and for that matter, importing DXF into SVG).

Of note, internally, Deepnest operates nearly directly on the SVG format (it makes a lot of assumptions based on the source being SVG).  It does a lot of polygon conversions, but always maintains a connection to the original SVG and when done generates the output SVG by linking it back to the original SVG to retain the properties.  That is real work to separate and likely not worth it.  Therefore, as indicated above, the best path is likely to find a good tool to do the conversion and integrate with it.

dorkmo, 2023-04-25:
Just as FYI, I believe this was the converter noted by jack
https://kabeja.sourceforge.net/