deepnest-io/Deepnest #5: Command Line Interface for Deepnest

bosd, 2023-04-16:
Requested in: https://github.com/Jack000/Deepnest/issues/101

Would it be possible to use deepnest with a CLI (or API) in order to automate the process ?

Found this in @odemakov 's fork..
`Start cli development`

https://github.com/odemakov/Deepnest/commit/94e72f775b8649467025aca5f9c764ce63bb4e61

Can someone look into this? Is this ready for an PR??

cmidgley, 2023-04-17:
Adding a cli itself is likely not too difficult (to start nesting given a set of files), but the code is structured such that it runs in a worker and never ends.  Likely the first effort would be logic, or at least policy (maximum number of runs, maximum time, something like that) to stop the runs so that the job could complete and save the resulting best SVG(s).

bosd, 2023-04-20:
> but the code is structured such that it runs in a worker and never ends. 

In one of the forks, I did see a stop implemented..

Well, there are bigger challenges for now :)

cmidgley, 2023-04-20:
If you find that fork again, post a reference to it here.  That would at least be a piece of the puzzle.

bosd, 2023-04-22:
I think i did see two implementations, for now I can only find some sort of timeout implementation.
In the fork of @mfussi 

https://github.com/mfussi/Deepnest

Quite a commit:
https://github.com/mfussi/Deepnest/commit/f8b7b3c3a53d52abbe3d9b64735f0dd64fe02a35

bosd, 2023-04-22:
@progand seems to be working recently on a CLI. https://github.com/progand/Deepnest

progand, 2023-04-24:
I tried to get rid of Electron, but SVG processing strongly depends on the browser.  JSDom has poor SVG support and doesn't help much. So I decided to hide the Electron window to be able to run deepnest as a console script.
The script stops executing once first solution is found.