deepnest-io/Deepnest #6: Update README for this community

cmidgley, 2023-04-17:
I have updated the old fork to reflect that it is now obsolete and this is the new fork to use for future work (therefore, do not merge with that fork else you get that readme update).

We should update this README to reflect the current state of the project (incl. removing my comments about not working on this), and add typical open-source things such as the goals of the community, how to participate, rules of engagement, and such.

bosd, 2023-04-20:
True, earlier I left everything unchaged.
I quickly changed some point that you mentioned. Yet, It  still needs the proper :heart:

Eskibrew, 2023-04-21:
Well done for taking steps to get this working again - I only came across it recently and did not realise that development had stalled.  I am interested in helping but I have also been trying to develop a FreeCAD workbench so I may not have much time available.  It looks like there is a lot to update to get it back to a point where things can move forward.  I'm interested to know what the plan will be.  Is there a discussion space or just here?

bosd, 2023-04-22:
Welcome @Eskibrew

> Is there a discussion space or just here?

For now it is just here. As all is still very new, there is still a lot todo.
Making the plan, roadmap, contributor guidelines, setting up ci/cd.

suchja, 2023-05-02:
Thanks for the fork of this project! While trying to get everything building on my machine I stumbled over the [Prerequisite section](https://github.com/deepnest-io/Deepnest/blob/master/README.md#prerequisites) in the Readme. The GitHub repo mentioned there is referring to @cmidgley. I guess it should be deepnest-io organization, correct?

abebeos, 2023-05-07:
> I guess it should be deepnest-io organization, correct?

Yes, fixed within https://github.com/deepnest-io/Deepnest/commit/085c4af4fb1545e140de292c33170aca6a76a104

abebeos, 2023-05-09:
@cmidgley , closing this, as it seems that the README is is fine now.

closed 2023-05-09 by abebeos