deepnest-io/Deepnest #19: Interesting forks & Related projects

bosd, 2023-04-22:
Deepnest with pdf support of @grvwebdev https://github.com/grvwebdev/deepnestweb

Deepnest fork with cli of @progand https://github.com/progand/Deepnest

# Related projects

**Cutnestprint**
Automatic solution for compact placement of images for printing. Made on Python, based on Django3. Allows two databases .sqlite3 to fast import data from the website. Uses the OpenCV computer vision library to build outlines of cuts. Uses Deepnest open source project for compact placement of parts
https://github.com/Dogthemachine/CutNestPrint

**Inkscape-nest**
https://github.com/Neon22/inkscape-nest

**SVG Nest Python fork**
[WIP?]
https://github.com/zhuyuanxiang/SVGnestPython

**SVGnest-batch**
GUI to process multiple files with SVGnest
https://github.com/stephanbogner/SVGnest-batch

dorkmo, 2023-04-25:
Not sure of the status but there was a FreeCAD nesting project:
https://wiki.freecad.org/Arch_Nest

This was also in my notes, mentions its archived now:
https://github.com/kallaballa/libnfporb

abebeos, 2023-05-09:
(issues #19, #16 #5 and #3 should be better moved over to Discussions, ideas)