deepnest-io/Deepnest #21: Building with VS2022 under Windows 11

suchja, 2023-05-02:
I'm trying to get this (as well as the InfernoEmbedded fork) to build for several hours now. Unfortunately the *node-gyp* is not working. Most likely it is not an issue at all with this repo, but with my configuration. However, as you guys figured out how to build, I hope someone can point me into the right direction. I'm willing to create some enhanced installation / build instructions, if you think it is required (I do ;).

I'm using Win11 (in a VM) and VS2022 (including the desktop C++ workload). I've installed NVM for windows and I am using node 8.17.0 as suggested. Also I finally installed pyenv with Python 2.7.18.

The problem I'm mainly facing is this:

> gyp ERR! find VS
gyp ERR! find VS msvs_version was set from command line or npm config
gyp ERR! find VS - looking for Visual Studio version 2022
gyp ERR! find VS running in VS Command Prompt, installation path is:
gyp ERR! find VS "C:\Program Files\Microsoft Visual Studio\2022\Community"
gyp ERR! find VS - will only use this version
gyp ERR! find VS unknown version "undefined" found at "C:\Program Files\Microsoft Visual Studio\2022\Community"
gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS looking for Visual Studio 2013
gyp ERR! find VS - not found
gyp ERR! find VS msvs_version does not match this VS Command Prompt or the
gyp ERR! find VS installation cannot be used.
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
gyp ERR! configure error
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\lernmoment\AppData\Roaming\nvm\v8.17.0\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
gyp ERR! stack     at findVisualStudio2013 (C:\Users\lernmoment\AppData\Roaming\nvm\v8.17.0\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16)
gyp ERR! stack     at regSearchKeys (C:\Users\lernmoment\AppData\Roaming\nvm\v8.17.0\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16)
gyp ERR! stack     at regGetValue (C:\Users\lernmoment\AppData\Roaming\nvm\v8.17.0\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7)
gyp ERR! stack     at C:\Users\lernmoment\AppData\Roaming\nvm\v8.17.0\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:288:5)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at maybeClose (internal/child_process.js:915:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
gyp ERR! System Windows_NT 10.0.22621
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\lernmoment\\AppData\\Roaming\\nvm\\v8.17.0\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\temp\deepnest-cmidgley
gyp ERR! node -v v8.17.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! deepnest@1.0.6-cm install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the deepnest@1.0.6-cm install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

I've added the following lines to my *.npmrc* and I would think, that it is working:
```
msbuild_path="C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\Community\\MSBuild\\Current\\Bin\\MSBuild.exe"
msvs_version=2022
```

However, it looks like it is not able to use my VS2022. Is somebody using VS2022 for build deepnest? Otherwise I would assume that node-gyp is simply not compatible with VS2022 (or at least not the old node-gyp version used here).

abebeos, 2023-05-04:
@suchja , may I suggest that we get initially your requirements straight?

* under Windows 11
  * => is this a MUST requirement, or can we fallback to Windows 10 here?

suchja, 2023-05-04:
@abebeos 

> may I suggest that we get initially your requirements straight?

Absolutely!

I currently have only Win11 for development. The release version should run on Win10. I wasn't aware that there are such differences. Is it because building the addon.cc involves the WinSDK?

Sorry, but I'm not aware of any other requirements. Ado you need to know something else?

abebeos, 2023-05-04:
> @abebeos
> 
> > may I suggest that we get initially your requirements straight?
(ensure to quote the original question)

> 
> Absolutely!
> 
> I currently have only Win11 for development. The release version should run on Win10. I wasn't aware that there are such differences. 

There are heavy differences across versions of the OS.

> Is it because building the addon.cc involves the WinSDK?

skipping (avoid questions during a RA)
 
> Sorry, but I'm not aware of any other requirements. Ado you need to know something else?

Not for now. I'll replay the installation on win10 pro.

In the meantime, you mentioned "I'm using Win11 (in a VM)", take a look if you can launch a Win10 VM.

abebeos, 2023-05-04:
@suchja , is the anonymous backer on bountysource you?

abebeos, 2023-05-04:
installation goes through on my Win10Pro machine.

We now have a few options:

- you try on a Win10 VM
- I try on a Win11 machine

I have however identified another thin detail:

> I've added the following lines to my _.npmrc_ and I would think, that it is working:
> 
> ```
> msbuild_path="C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\Community\\MSBuild\\Current\\Bin\\MSBuild.exe"
> msvs_version=2022
> ```

Try to reinstall VS2022 within

`C:\Program Files\Microsoft Visual Studio\2022`

**NOT LIKE NOW WITHIN**

`C:\Program Files (x86)\Microsoft Visual Studio\2022`

and retry.

bosd, 2023-05-04:
My tech skills are not on par for this. But maybe I can give some pointers..

@suchja 
> I'm trying to get this (as well as the InfernoEmbedded fork) to build for several hours now.

The code in this repo is from @cmidgley, which also has containts the commits of the InfernoEmbedded fork.
Correct me if I'm wrong.

> However, as you guys figured out how to build
I think this is the main issue now with this code.
(Personally I don't have the knowdledge/ did not even make an attempt)

Another fork which might be wortwhile looking into is this one.
Some dependencies are updated. Also supports a more recent python version (or not??)
https://github.com/9swampy/Deepnest/tree/develop

The changes from that repo have not been commited here.
There is an docker image based on this one. Which some people had some success with.

abebeos, 2023-05-04:
@suchja, I have reproduced your error locally. I'll let you know once I've found a fix.

abebeos, 2023-05-04:
@suchja , I can eliminate the error (you have) on my side simply by falling back to VS2019

Can you try this:

### Build with VS2019

```
# unset VCINSTALLDIR
set VCINSTALLDIR=

npm config set msvs_version 2019

npm install

npm run w:build

npm run w:start
```

### VC2022
>Otherwise I would assume that node-gyp is simply not compatible with VS2022 (or at least not the old node-gyp version used here).

after doing some research, this seems to be true. Not sure how trivial it is to update (you should avoid this, use the existent code of this project until it is updated officially)

abebeos, 2023-05-04:
> Sorry, but I'm not aware of any other requirements. Ado you need to know something else?

Is `VC2022` a MUST requirement, or can you fall-back to VC2019?

If VC2019 is ok, the the issue is solved (I would update the doc here a little bit to make things clearer, thus others do not fall into the same trap)

abebeos, 2023-05-05:
draft revision of the build instructions:

- https://github.com/deepnest-io/Deepnest/issues/23

cmidgley, 2023-05-05:
I attempted to upgrade to the latest everything earlier (including the latest Node and therefore node-gyp), but there were too many changes to Electron involved so I moved it as far forward as possible without breaking that.  Somebody experienced with Electron likely could do those changes efficiently but it's not my expertise.  There may be more involved, but that seemed to be the bulk of the lift.

suchja, 2023-05-05:
Thank you all for your immense effort and the high quality and fast feedback! I'll start from scratch with a new VM and give the different provided options a try. Hope that I'll be able to do so on this weekend.

abebeos, 2023-05-05:
Hey, @suchja, i got involved here in context of the bounty, see https://github.com/deepnest-io/Deepnest/issues/21#issuecomment-1534328266.

Ignoring my concrete questions and results is not the way to go.

* Solution https://github.com/deepnest-io/Deepnest/issues/21#issuecomment-1534730830
* question https://github.com/deepnest-io/Deepnest/issues/21#issuecomment-1535076621
* docs https://github.com/deepnest-io/Deepnest/issues/21#issuecomment-1535715715

You do not need a new vm.

abebeos, 2023-05-05:
@cmidgley , I've added your info re updating to

- #24

abebeos, 2023-05-06:
I've added a basic CI here, and it builds fine, just by falling back to a windows-2019 runner (uses VS2019).

https://github.com/deepnest-io/Deepnest/actions/runs/4899908971/jobs/8750207768

So, @suchja, this issue is solved, once you've verified on your side, please close thus I can process the bounty and go on with the other tasks here.

abebeos, 2023-05-07:
Did some additional work to get the project running, see APEX issues and the MAJOR issues/sub-issues:

- https://github.com/deepnest-io/Deepnest/issues/35

abebeos, 2023-05-08:
Closing, in order to speedup things and go on with:

- https://github.com/deepnest-io/Deepnest/issues/31

abebeos, 2023-09-26:
@suchja , hi there!

As you possibly know, bountysource stopped paying out bounty-workers since early this year

https://github.com/bountysource/core/issues/1586

The bountysource app is down since mid Sep. 2023

https://app.bountysource.com/issues/120217906-building-with-vs2022-under-windows-11

### Paypal Buyer Protection

Though you got the service from me, I did not receive the payment, bountysource has kept the money.

You are now eligible for paypal buyer protection (others have gone successfully this path) :

https://github.com/bountysource/core/issues/1586#issuecomment-1732668211

If you could do this, and pay me directly, you could keep the 10% for your efforts.

Thank you.

closed 2024-06-15 by abebeos