deepnest-io/Deepnest #24: Update Versions

abebeos, 2023-05-05:
(this is an incremental process)

- [ ] Disable renovate for now
- [ ] Identify what can be updated easily (most possibly trial&error)
- [ ] (...)

ruanhuishi, 2023-05-16:
I ran the project and selected SVG; But I didn't get the NEST calculation results！

Can you assist in solving the problem?
[image]

abebeos, 2023-05-16:
> Can you assist in solving the problem?

Please see: https://github.com/deepnest-io/Deepnest/discussions/80#discussioncomment-5914410

abebeos, 2023-05-17:
@dorkmo , closing this as fulfilled (the electron update is out of scope of this chunk)

- follow-up: #83

closed 2023-05-17 by abebeos