deepnest-io/Deepnest #35: [APEX] Work Plan

abebeos, 2023-05-07:
(please do not edit for as long as it is assigned)

- [ ] #31 
- [ ] #34
- [ ] #24

r-bt, 2023-08-17:
Hey, just wondering if there is a branch with the update to electron 12? I'm interested in helping to get this to work on macOS but right not it's not compiling since theres (obviously) no version of electron for ARM macbooks

abebeos, 2023-08-22:
@r-bt, I'll take a look at the status later and inform you here.

abebeos, 2023-08-22:
> Hey, just wondering if there is a branch with the update to electron 12?

There is no public branch, I finalized the work locally (end of May 2023), despite not getting the budget.

Take a look at the updated issue text and at:

- https://github.com/deepnest-io/Deepnest/issues/109

After this, we would most possibly go for the stabilization on linux/mac (if there are volunteers to do the testing/reporting/possibly some tiny coding tasks).

bosd, 2023-08-23:
> I'm interested in helping to get this to work on macOS but right not it's not compiling since theres (obviously) no version of electron for ARM macbooks.

I have absolutly no experience with macos.
But maybe this is usefull. I got it to build version 1.1.4 on Linux using the script in my fork.
https://github.com/bosd/Deepnest/tree/path

bosd, 2023-08-23:
> (if there are volunteers to do the testing/reporting/possibly some tiny coding tasks).

Count me in for this! 😄

abebeos, 2023-08-28:
> @abebeos Hey, just wondering if there is a branch with the update to electron 12?

@r-bt , v1.1.5 is out, see tags/releases