deepnest-io/Deepnest #40: Clarify Naming

abebeos, 2023-05-09:


bosd, 2023-05-09:
> the deepnest-io github organization
name should be used only if deepnest.io domain is under control

The domain is not under my (or anyone else's) control in this group.
The original author of deepnest is invited to join this organization. Yet, he's unresponsive and not very active on github anymore.

Propably a new name for this product and repo is better.
Suggestions anyone??

bosd, 2023-05-09:
> Possibly change for now the github organization to "deepnest" (strip the "-io" part), in order to avoid confusion with deepnest.io.

I'm open for a name change. When I opened this repo and organization a few weeks ago, it was not possible to use deepnest as a name.
At the time i tried some variants like `deepnest-x` `deepnest-ng` and whatever. 
Eventually settling on the current one deepnest-io.
Already noted that it could lead to confusion.

Main reason that I went for a name which still included deepnest. Is because the name is still very popular.
Yet it is sad that a lot of people go the https://github.com/Jack000/Deepnest/ repo and are communicating on a repo which is no longer active / maintained.

dorkmo, 2023-05-10:
i agree it might be polite and proper to do a name change. Probably something like opennest or librenest ?

abebeos, 2023-05-10:
> opennest

Cadnest, partnest,