deepnest-io/Deepnest #61: [FEEDBACK] I need Test Data Files

abebeos, 2023-05-13:
I need some simple medium sized files in order to run tests.

dorkmo, 2023-05-14:
heres some stuff i've worked with, mostly rectangular though, so maybe might need some other examples with curves
https://github.com/OpenSourceEcology/CEB-Press/tree/master/v1708P/xport2023

abebeos, 2023-05-14:
> mostly rectangular though

All good, this is mostly about testing the electron/js upgrade stuff.

The important thing is that those are validated with Deepnest, as I've currently problems: 

- #65

abebeos, 2023-05-14:
I've used the SVG files included in the distribution https://deepnest.io/#download.

The nesting works with the updated versions (I have reached node 14).

### Important

I am new to this software. I do not know:

- what currently works
- what does not work

So, possibly this should be documented clearly, e.g. like:

* importing DXF does not work
* importing SVG works
* etc.

abebeos, 2023-05-15:
I have produced such data by simply exporting a dxf file with the original `1.0.5` version.

This DXF can be imported into our updated version `1.1.1`, so all is good, no regressions.

closed 2023-05-15 by abebeos