deepnest-io/Deepnest #100: No preview screen when nesting

concifederico, 2023-06-03:
Ubuntu 22.04 no preview screen, infinite nesting loop without ending:

[image]

aardwolfweb, 2023-06-09:
Seeing the same behavior on KDE Neon 5.27. (Ubuntu derived)

thejcannon, 2023-06-16:
FWIW I see this in terminal:

```console
$ npm run start

> deepnest@1.1.4 start
> electron .

(node:33901) UnhandledPromiseRejectionWarning: ReferenceError: createWindow is not defined
    at /home/josh/Downloads/Deepnest/main.js:79:5
(node:33901) UnhandledPromiseRejectionWarning: ReferenceError: createWindow is not defined
    at /home/josh/Downloads/Deepnest/main.js:79:5
(node:33901) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:33901) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:33901) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:33901) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
starting background!
stopped! [
  BrowserWindow {
    setBounds: [Function],
    _events: [Object: null prototype] {
      blur: [Function],
      focus: [Function],
      show: [Function: visibilityChanged],
      hide: [Function: visibilityChanged],
      minimize: [Function: visibilityChanged],
      maximize: [Function: visibilityChanged],
      restore: [Function: visibilityChanged],
      'ready-to-show': [Function]
    },
    _eventsCount: 8,
    devToolsWebContents: [Getter]
  }
]
```

linaflorez, 2023-07-06:
I'm dealing with this on my mac as well. Has anyone found a solution?

tisaconundrum2, 2023-08-13:
After working at this, I'm reasonably confident that this an issue with these two files. Something to do with the updates not being passed to the display call back. Still digging through it.

[image]

bosd, 2023-08-17:
One difference between the Linux and windows is in the log.
On linux system the log is empty.
On Windows when it is corretly nesting and previewing it prints:
`starting background!`
(and keeps on printing those lines)

tisaconundrum2, 2023-08-17:
Time to dockerize the whole thing 😜

bosd, 2023-08-18:
> Really the whole thing needs a rework and follow best practices of Electron App design

Who can do such thing?

abebeos, 2023-08-22:
> Who can do such thing?

no need to dig into bugs on different OS than windows while the electron port to newer versions is still active.

remember that work was aborted due to lack of funds:

https://github.com/deepnest-io/Deepnest/issues/35

abebeos, 2023-08-24:
(stable working windows version should be updated first to modern/latest versions of tools, then the validation on linux/mac should happen)

Having said the above:

- if you go the "unsupported" track, provide very concise reproduction info (all version, including compilers etc.)
- the problem is most possibly very(!) old versions of libraries, and it can very soon become a major task in itself to fix it.