deepnest-io/Deepnest #103: Deepnest won't start with ubuntu 22.04

thierry7100, 2023-06-23:
The AppImage won't start properly, first time it proposes to integrate the software in the system. This is working fine. But software won't start. When launched from terminal, I got a segmentation fault message

**To Reproduce**
Steps to reproduce the behavior:
1. launch a terminal.
2. Goto  the folder with Deepnest
3. Launch Deepnest from terminal : ~/Applications/Deepnest-1.0.5-linux$ ./Deepnest-1.0.5-x86_64.AppImage 

**Expected behavior**
Software start

**Screenshots**
thierry@thierry-UX410UAR:~/Applications/Deepnest-1.0.5-linux$ ./Deepnest-1.0.5-x86_64.AppImage 
Segmentation error (core dumped)

**Desktop (please complete the following information):**
 - Ubuntu 22.04 (without Wayland if it helps)
 
**Additional context**
Add any other context about the problem here.

concifederico, 2023-06-25:
Same here, however the 1.05 version AppImage source has another git --> https://github.com/Jack000/Deepnest
There is no AppImage of this fork yet, maintainer claims this is not a priority.
Regards,

Federico

abebeos, 2023-07-17:
Hi there. This fork here was not yet enabled on linux, only windows was verified, see packages.

closed 2023-08-24 by abebeos

abebeos, 2023-09-16:
should work with https://github.com/deepnest-io/Deepnest/releases/tag/v1.1.6