deepnest-io/Deepnest #104: Deepnest doesn´t run in Linux Mint

MautaEve, 2023-07-27:
Hi!
after unzip and give permission to execute, i double click and nothing happen. 
with the console i have this answer: 

Segmentation fault (core dumped)

what should I do? Is there a way to fix this? Do you need more info?

aardwolfweb, 2023-07-28:
This current version is only supported on Windows. The older version for Linux uses some older versions of libraries that are out of date. I ended up running it in a Windows 10 virtual machine.

bosd, 2023-08-15:
@MautaEve 
Can you provide more details about the errors?
Does the main window open?

Did you try to build it from source??

I've been having a small bit of succes with building on Fedora for steps see #107
I've modded the script section of package.json to this:

```
  "scripts": {
    "start": "electron .",
    "configure": "node-gyp configure --release",
    "build-all": "node-gyp rebuild && mkdir minkowski >nul 2>&1|echo .>nul && npm run build",
    "build": "mkdir minkowski >nul 2>&1|echo .>nul && mkdir minkowski\/Release >nul 2>&1|echo .>nul && .\/node_modules\/.bin\/electron-rebuild && npm run copy",
    "clean": "rmdir /s /q build dist >nul 2>&1|echo .>nul",
    "clean-all": "rmdir /s /q build dist node_modules minkowski bin >nul 2>&1|echo .>nul",
    "dist": ".\\node_modules\\.bin\\electron-packager . deepnest-v%npm_package_version% --platform=win32 --arch=x64 --overwrite",
    "dist-all": "npm run clean-all && npm install && npm run build-all && npm run dist",
    "copy": "cp -r .\/build\/Release .\/minkowski\/Release\/"
  },
```

I've got the main window opening. Loading dxf works. But nesting is not working yet.