deepnest-io/Deepnest #105: Electron failed to install correctly

Chen1098, 2023-08-11:
I can't complete the step "npm run start" on windows 11 visual studio 22. PowerShell returned the following
`PS C:\Users\drwen\Desktop\Project\NewDeepnest\Deepnest-master> npm run start

> deepnest@1.1.4 start
> electron .

C:\Users\drwen\Desktop\Project\NewDeepnest\Deepnest-master\node_modules\electron\index.js:14
    throw new Error('Electron failed to install correctly, please delete node_modules/electron and try installing again')
    ^

Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at getElectronPath (C:\Users\drwen\Desktop\Project\NewDeepnest\Deepnest-master\node_modules\electron\index.js:14:11)
    at Object.<anonymous> (C:\Users\drwen\Desktop\Project\NewDeepnest\Deepnest-master\node_modules\electron\index.js:18:18)
    at Module._compile (node:internal/modules/cjs/loader:1149:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1203:10)
    at Module.load (node:internal/modules/cjs/loader:1027:32)
    at Module._load (node:internal/modules/cjs/loader:868:12)
    at Module.require (node:internal/modules/cjs/loader:1051:19)
    at require (node:internal/modules/cjs/helpers:103:18)
    at Object.<anonymous> (C:\Users\drwen\Desktop\Project\NewDeepnest\Deepnest-master\node_modules\electron\cli.js:3:16)
    at Module._compile (node:internal/modules/cjs/loader:1149:14)

Node.js v18.10.0
NativeCommandExitException: Program "npm.cmd" ended with non-zero exit code: 1.`
I have deleted the node_modules and reinstalled it many times with no effect whatsoever and after googling this problem, a website told me to delete ""electron": "^6.1.12"," in package.json and retry, which didn't work either.
Hope I can get a reply soon P.s. I am willing to share my gpt 4 account to whoever solved this problem lol
Best , Chen

Chen1098, 2023-08-11:
P.S. I am in China and am using a VPN for all this downloading (maybe that is why??)

Chen1098, 2023-08-11:
[image]
I think I have found the problem here.....

bosd, 2023-08-11:
The information if cmidgley might be outdated. As abesos continued development based on cmidgley's work.

Chen1098, 2023-08-12:
> The information if cmidgley might be outdated. As abesos continued development based on cmidgley's work.

who is abesos? I don't think i can find him (her) on github. Mind sending me his (her) link? Thanks

bosd, 2023-08-12:
Sorry, it is [abebeos](https://github.com/deepnest-io/Deepnest/commits?author=abebeos)

abebeos, 2023-08-23:
@Chen1098, you can try:

- another node.js version
- a build with another compiler ("or, as an administrator via npm install --global windows-build-tools (older VS version)")
- or follow this suggestion here (which is possibly related)
  - https://github.com/electron/electron/issues/8466#issuecomment-274380021