deepnest-io/Deepnest #107: [Fedora] Build fails Xcopy

bosd, 2023-08-15:
**Describe the bug**
Unable to build on Fedora Linux. Xcopy command not found.
The xcopy command is windows specific.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the Repo
2. `cd Deepnest` 
3. `sudo dnf groupinstall "Development Tools" "Development Libraries"`
4. `sudo dnf install make automake gcc gcc-c++ kernel-devel`
5. `npm install`
6. `npm run build`

**Expected behavior**
Deepnest is able to build. (I know other platforms then windows are not yet supported)

**Screenshots**
```

[bosd@fedora Deepnest]$ npm run build

> deepnest@1.1.4 build
> mkdir minkowski >nul 2>&1|echo .>nul && mkdir minkowski/Release >nul 2>&1|echo .>nul && ./node_modules/.bin/electron-rebuild && npm run copy

✔ Rebuild Complete

> deepnest@1.1.4 copy
> xcopy /s /y .\build\Release .\minkowski\Release

sh: line 1: xcopy: command not found
```

**Desktop (please complete the following information):**
 - OS: Fedora 38
 - Deepnest 1.1.4
-  npm 8.19.2

**Additional context**
Propably better to use some form of platform independent copy method.

bosd, 2023-08-15:
xcopy seems Windows dependent.
CP seems Unix dependent.

Maybe it is possible to implement [fs.copy](https://nodejs.org/api/fs.html#fscpsrc-dest-options-callback).
To be platform independent.
source: https://stackoverflow.com/a/76375643

tisaconundrum2, 2023-08-15:
yes, this is correct. `xcopy` is a windows based command. If you're using `.ps1` file, know that that is a dedicated powershell script used for Windows.

abebeos, 2023-08-22:
> Unable to build on Fedora Linux.

@bosd The build is currently only validated on Windows.

bosd, 2023-08-23:
Maybe not the highest priority to make the build script platform independent.
I've had succes building it on linux (fedora 38) with my the linux version of the script in my fork.
https://github.com/bosd/Deepnest/tree/path

abebeos, 2023-08-24:
> Maybe not the highest priority to make the build script platform independent.

It is highly important that you do not "break" the development flow, we need to focus on this here:

- https://github.com/deepnest-io/Deepnest/issues/109

As for solving the xcopy issue, it is trivial, e.g. via this:

https://stackoverflow.com/a/59823713/21521485

(but no one should work on the electron 6 version, when a electron 12 version is available)

commit ad496eda (2023-08-25): introduce cross-platform build, fix #107
commit b19de4cc (2023-08-25): initial ubuntu support, fixes #107 (#112)
commit ad7a4d4a (2023-08-26): initial ubuntu support, fixes #107 (#112)
commit f62b9988 (2023-09-26): initial ubuntu support, fixes #107 (#112)
closed 2023-08-25 by abebeos