deepnest-io/Deepnest #108: ReferenceError: createWindow is not defined

bosd, 2023-08-15:
**Describe the bug**
The nest window stays blank.
There is an error message in the log.

**To Reproduce**
Steps to reproduce the behavior:
1. `npm run start`

**Expected behavior**
Deepnest to start normally and nest

**Log**
```
(node:645549) UnhandledPromiseRejectionWarning: ReferenceError: createWindow is not defined
    at /home/data/git/Deepnest/main.js:79:5
(node:645549) UnhandledPromiseRejectionWarning: ReferenceError: createWindow is not defined
    at /home/data/git/Deepnest/main.js:79:5
```

**Desktop (please complete the following information):**
 - OS: Fedora 
- Deepnest 1.1.4
- npm 8.19.2

**Additional context**
- [might be related](https://stackoverflow.com/questions/63070777/mainwindow-is-not-defined-electron-js)

bosd, 2023-08-15:
Just noticed the log output was also mentioned in #100

abebeos, 2023-08-22:
We are currently "windows only". Could check for linux (and possibly Mac) after introducing 1.05 (electron 12)

abebeos, 2023-08-26:
@bosd linux currently unsupported, moving over to discussions.