deepnest-io/Deepnest #109: Make v1.1.5 public (updated to electron 12)

abebeos, 2023-08-22:
Copy of private messaged to funders (end of May):

i've reached:

- electron 12.2.3 (this is enough, for now)
- structural refactoring of the C++ Unit (much simpler, less dependencies)
  - ready for code refactoring / development
- The budget for publishing v1.1.5 sources on the project repo is $2500

(to clarify: v1.1.5 is ready)

v1.1.5 is the foundation to start coding work (like DXF import/export, JPG import/export, Linux build, functional additions/changes, JS/C++ refactoring etc.)

### Going on

If your willing to contribute to the budget for this release, just add a comment here.

bosd, 2023-08-26:
I've just released the funds for version 1.1.15  
But we need more contributor's and sponsors!

abebeos, 2023-08-27:
> I've just released the funds for version 1.1.15

1.1.5, received $500

> But we need more contributor's and sponsors!

v1.1.5 + linux should open up for more contributors/sponsors. Kind of a dead-lock.

abebeos, 2023-08-27:
@bosd . published upfront to speed things up, see #122