deepnest-io/Deepnest #119: deepnest.io still links to original repository

iliis, 2023-08-27:
I don't know who manages/owns https://deepnest.io/, but it still links to https://github.com/Jack000/Deepnest instead of this repo.
It also provides and older version (1.0.5) than what's available [here](https://github.com/deepnest-io/Deepnest/releases).

iliis, 2023-08-27:
I just saw https://github.com/deepnest-io/Deepnest/discussions/53, so it seems the site isn't controlled by this project. Sorry for the spam.

abebeos, 2023-08-27:
> I just saw #53, so it seems the site isn't controlled by this project. Sorry for the spam.

All good, your issue is valid

@bosd , we should drop the "-io" part for now to avoid confusion (see https://github.com/deepnest-io/Deepnest/discussions/53#discussioncomment-5874898)

bosd, 2023-08-27:
> @bosd , we should drop the "-io" part for now to avoid confusion (see [#53 (comment)](https://github.com/deepnest-io/Deepnest/discussions/53#discussioncomment-5874898))

The "-io" cannot be dropped as the name deepnest is not available on github.

Maybe it is time, for a complete rebrand of this thing.
Then we can create a own website, to for once and for all close any possible confusion.

Yet, there is a bigger challenge with this project.
Which is the lack of contributors, maintainers & sponsors.
(see #109)

Just thinking out loud... Maybe it is viable to switch to an open core model?
Where the paid version will have advanced planning features which could be integrated into a clients erp/manufacturing system.  
( I know, should be seperate discussion, but closely tied together. Better rebrand once )

abebeos, 2023-08-27:
> The "-io" cannot be dropped as the name deepnest is not available on github.

deepnest-net, deepnest-alt, deepnest-fork, deepnest-temp, deepnest-new etc.

> I know, should be seperate discussion

yes.

abebeos, 2023-08-28:
@bosd , @iliis , I've changed the organization description to 

> Community-based organization (we do not own deepnest.io) that maintains a deepnest fork

and added a readme, see https://github.com/deepnest-io

So we can keep the deepnest-io github organization name for now.

closed 2023-08-28 by abebeos