deepnest-io/Deepnest #120: SVG import doesn't clean up polyline

iliis, 2023-08-27:
Importing an SVG with paths described by the [polyline tag](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/polyline) will result in open line segments instead of being merged into a closed path.

**To Reproduce**
Steps to reproduce the behavior:

1. Download the following file (right click and 'Save Image As...'). It contains three squares (from left to right):
    1. A rectangle [(svg:rect)](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/rect)
    2. A square made out of four paths [(svg:path)](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/path)
    3. A square made out of four polylines [(svg:polyline)](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/polyline).
    [image]
4. Import it in Deepnest

**Expected Behavior**
Three squares, as individual shapes.

**Observed Behavior**
Only the first two squares are correctly imported, the third one is ignored and shown in red.
[image]

**Desktop:**
 - OS: Windows 10 Enterprise (10.0.19045 Build 19045)
 - Deepnest Version: [v1.1.4](https://github.com/deepnest-io/Deepnest/releases/tag/v1.1.4)

**Additional context**
This issue came up when trying to import an SVG exported by the SVG exporter from [OnShape](https://www.onshape.com/). Exporting DXF seems to run into the same problem.
I can work around the issue with Inkscape by selecting everything and applying 'Path' -> 'Object to Path'. However, on a production file this takes a really long time and isn't really practical.

abebeos, 2023-08-27:
Thank you for your thorough report. Development has just restarted after a long summer-break. We are currently stuck with funding (a major sponsor went silent), see https://github.com/deepnest-io/Deepnest/issues/109. After publishing version 1.1.5, work on detail level (like this issue) can happen.

abebeos, 2023-08-27:
@bosd can you please verify the bug and apply the https://github.com/deepnest-io/Deepnest/labels/confirmed label once done?

abebeos, 2023-08-27:
> Importing an SVG with paths described by the [polyline tag](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/polyline) will

Are you sure that this is a commonly used construct? According to:

> Typically a polyline is used to create open shapes as the last point doesn't have to be connected to the first point. For closed shapes see thehttps://developer.mozilla.org/en-US/docs/Web/SVG/Element/polygon element.

"polygon" should be used.

### Issue

The issue seems to be:

=> Deepnest does not recognize shapes formed by multiple polylines

(which looks to me like a non-issue, or special-case/non-standard functionality)

iliis, 2023-08-27:
I do not know how common this is, but at least my CAD software exports a lot of shapes as polylines.
Here's an SVG straight from an OnShape drawing:

[image]

This actually loads fine, but has an unwanted border and thus loads as one single piece in Deepnest:

[image]

When I ungroup the elements and remove the border in Inkscape and save it as (plain) SVG:
[image]

Then it looks like this in Deepnest:
[image]

Which is interesting, as some shapes are imported correctly, even though they are segmented polylines too. So maybe Deepnest tries to clean up the polylines but somehow fails for some? I tried various settings for the endpoint tolerance, but this didn't seem to have any effect.

---

I agree that maybe the CAD software should export with a different tag. However, this is a *2D drawing* I'm exporting (projected from the 3D geometry), which in general can contain open open shapes (e.g. arrows or dimensions), so maybe that's why they are exporting everything as polylines.

I would argue this is still an issue though, at least for myself and following Postel's law ("be conservative in what you do, be liberal in what you accept from others"). If I have time I'll have a look at the code, but I'm not used to JavaScript.

abebeos, 2023-08-27:
> my CAD software

Which is it? Maybe it has a configuration to tweak the output?

> Postel's law

No postel's law here. Just the brutal rality of "lack of sponsors"...

iliis, 2023-08-27:
> > my CAD software
> 
> Which is it? Maybe it has a configuration to tweak the output?

https://www.onshape.com/ and no, unfortunately it doesn't.

> No postel's law here. Just the brutal reality of "lack of sponsors"...

I can spare a couple bucks, but nothing major. But I'm also not expecting you to do anything, just thought I'd report my findings.

abebeos, 2023-08-27:
> I can spare a couple bucks, but nothing major.

A few "nothing major" (and some elasticity from my side) can do it.

I've invited you to a private repo with sponsor info.

> But I'm also not expecting you to do anything, just thought I'd report my findings.

I'll at minimum document the problem domain, thus it becomes easier to solve it (for me, later, or for others).

> I tried various settings for the endpoint tolerance, but this didn't seem to have any effect.

I had myself import failures with some test-data, after accidentally changing the deepnest configuration. So possilby play with other config options, too. Not only  endpoint tolerance

abebeos, 2023-08-27:
- off-topic-

> https://www.onshape.com/

is the free version workable?

abebeos, 2023-08-29:
> Which is interesting, as some shapes are imported correctly

it seems those imported contain all  circles (full/partial)

abebeos, 2023-08-30:
see https://github.com/deepnest-io/Deepnest/issues/124#issuecomment-1699113209

abebeos, 2023-09-03:
fixed in 1.1.9

abebeos, 2023-09-05:
@iliis OnShape export can now be directly imported (1.1.10)

abebeos, 2023-09-08:
@iliis , please see #130

iliis, 2023-09-14:
Thank you very much for woking on this! I've [replied at the other issue](https://github.com/deepnest-io/Deepnest/issues/130#issuecomment-1719422510).

abebeos, 2023-09-26:
@iliis , I'll reopen this as i'll publish 1.2.0 in the next days.

Funding deficit is anyways not your fault (you just came into the cross-fire).

abebeos, 2023-09-30:
@iliis , could you check this on your side to confirm it works?

commit 57fa7ba9 (2023-09-26): allow further processing of 2-point polylins

fixes #120
closed 2024-06-15 by abebeos