deepnest-io/Deepnest #128: v1.1.9 introduces linux support & polyline import (e.g. OnShape)

abebeos, 2023-09-03:
- unpublished
- #109

abebeos, 2023-09-05:
@dorkmo , @bosd , @iliis

I guess I halt development for now. All looks very good (e.g. direct OnShape/Inkscape import) - but beeing totally alone within this project is really exhausting.

bosd, 2023-09-06:
Maybe more contributors will follow. Like in #127 
This repo is relative fresh.

abebeos, 2023-09-06:
A few random contributions/ideas do not help - we saw what happened with the many forks

What is needed is 

- funding (cash or agreement)
- presence (people here, including you, tend to keep silence)
  - this is depressing/demotivating

bosd, 2023-09-06:
> A few random contributions/ideas do not help - we saw what happened with the many forks
> 
> What is needed is 
> 
> - funding (cash or agreement)
Very clear, hope more sponsors to come.
(Or time to think about a business model to make it at least break even.)

> - presence (people here, including you, tend to keep silence)
Not really clear what you want with this. The sparse spare time I have a available I am contributing to this ( and other open source projects).

>   - this is depressing/demotivating

Let's hope that more people find their way to this repo. So there are more shoulders to carry this project. 

I'm grateful for the push you gave to this project.

abebeos, 2023-09-06:
> So there are more shoulders to carry this project.

This is not about more(!) people, but about the functioning of the existent people (or those that appear now and then).

I ask for something, then I delete it again after two days, as no one replies.

I absolutely hate it when people go silent (instead of keeping the unwritten 24h reply rule).

As things stand, there are currently 2 people here:

- me, doing technical stuff
- you (@bosd), who should do things like getting more people on board, trying to get more funds etc.

You keeping silence for days does simply not help.

Anyways, I'll see to it to find a workaround to all this (but i'll not again publish stuff upfront...)

bosd, 2023-09-07:
Thanks for this clarifaction of expectations.

I understand it is desirable to have a quick response time. Yet replys within a couple of days is not an realistic expectation to an side project as this.
I see bigger projects with more contributors where it takes months to get feedback on a issue/pr.

Yet, your comment is noted. I try to improve. 😄

abebeos, 2023-09-07:
> I understand it is desirable to have a quick response time.

It is a must(!) requirement, non-disputable.

Pros do not even spend one second talking about this.

> Yet replys within a couple of days is not an realistic expectation to an side project as this.

It is not realistic to expect that a project-setup, many members and a few issues will solve things.

It is not realistic to expect to provide a limited budget, and leave the developer witout any support for days (if not weeks).

Actually this is ridiculous.

> I see bigger projects with more contributors where it takes months to get feedback on a issue/pr.

That's another thing. If you spend money on a developer, you (rationally) provide support, to get the most out of the developer.

> Yet, your comment is noted. I try to improve. 

You don't have to. It is your mentality that has to change. This is nearly impossible (at least not as fast as necessary).

I have for now changed the inactive "Owners" to "Members" (you should downgrade yourself). (Alternative would be for me to open a new repo to have more control, but would like to avoid this)

I will then follow-up with an Issue re "obfuscation", so we can have the working product out without handing out sources (can be done later if more funding comes in).

abebeos, 2023-09-07:
@bosd , see

- #129

abebeos, 2023-09-08:
Closing this in favor of 

- #130

closed 2023-09-08 by abebeos