deepnest-io/Deepnest #130: Donate to Dev Fund and Pick Available Functionality/Fixes for Inclusion

abebeos, 2023-09-08:
You can pick one to include to the repo by donating $500 to the development fund.

(mixed calculation, one was more work, the other is less)

Functionality/Fix is available locally:

- [ ] enable direct import from OnShape (remove rectangle)
- [ ] enable polyline-centric svg (OnShape/Inkscape)
- [ ] enable preview-screen on linux
- [ ] electron 13

abebeos, 2023-09-10:
@bosd , let's go "bazar":

send $500 for those 2 items:

* enable preview-screen on linux
* electron 13

this should simplify enabling the software for mac-os, too.

I'll then clean-up the project & docu a bit and prepare it for contributiors, ideally deepnest could the "self-heal" (users sending PRs in, to fix smaller issues).

For the case that I choose to continue on an "founders only" non-public version, you and @dorkmo will get access to it (as the only 2 who contributed financially).

bosd, 2023-09-11:
@abebeos Wanted to respond to you in private. Yet the private channel seems gone.

abebeos, 2023-09-11:
> Yet the private channel seems gone

Yes, me shredding. Just use email.

iliis, 2023-09-14:
Thank you for actually working on this! I did not expect such a response. And I'm sorry for not replying earlier, I was on vacation the last couple of weeks. I understand and sympathize with your frustration, but I feel you have unrealistic expectations here. A small open source project is not going to be a self-sustaining business.
The OSS projects I'm used to are usually small hobby projects people work on for fun and self-need, where works happens whenever there's time and motivation, often with weeks and months or even years between changes.
I do not expect any work from you or anybody else (apart from maybe the most basic of administration as the repo owner) and if you'd like to move this project into a more commercial direction that's up to you. I can donate 50$-100$ for hosting costs and suchlike and considering you've already worked on my issue, but paying for the development costs of a small tool I'm only going to use a handful of times for my own private side projects is way over my budget, I'm sorry.

abebeos, 2023-09-14:
> I feel you have unrealistic expectations here.

Your 'feelings' are confused, unfounded and... unrealistic.

I'm out here, this started fine with decent bounties/budgets, but I'll now shut all effort down.

> I can donate 50$-100$ for hosting costs and suchlike and considering you've already worked on my issue,

Nah, spend those on some kiddo-devs.

iliis, 2023-09-14:
Alright then, good luck with this project.