deepnest-io/Deepnest #133: SVG export viewbox height is wrong

DBGHacc, 2023-10-01:
**Describe the bug**
I exported an SVG of a layout. The sheet size was 5000 x 1500 mm. In the exported .SVG file the viewbox was width="5000mm" and height="1650.0000000000002mm". Where did the extra 150 mm come from?

It seems to calculate with the correct size though. When i changed the height in inkscape, the positioning of the parts seemed correct compared to the preview.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a sheet and parts.
2. Start nest.
3. Export to an .SVG file
4. Open in SVG editor or text editor and check the size of viewbox-item.

**Expected behavior**
The viewbox should be the same as the main sheet.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.20

abebeos, 2023-10-01:
Not sure why the original author added 10% to the height.

https://github.com/deepnest-io/Deepnest/blob/d023b9dde11bbe294ca75ad76487bdf58529ccb9/main/index.html#L1199-L1201

Be aware that it looks like nobody complaint about this, so not 100% sure if it is a bug.

You could try to simply change the code and try if it still works (no need to recompile, you can `npm run start` from the source-code).

Otherwise, if you want this to get verified and included in a new version, consider a contribution to the dev-fund, and I'll possibly start again work on code-level (currently just monitoring issues here).

Dexus, 2024-12-09:
Is not a bug. It's more a separation between multiple sheets.