deepnest-io/Deepnest #135: CLI Option

davewatson91, 2023-10-03:
I've had a dig, but not been able to come up with anything. 

Is there any way to run this as a CLI tool? Looking to interface with some other processes in-place. 

Thanks!

abebeos, 2023-10-03:
See https://github.com/deepnest-io/Deepnest/discussions/45 at the end.

davewatson91, 2023-10-03:
> See #45 at the end.

Thanks @abebeos. I've just had a look at them, but can't see anything explicitly supporting a CLI. I can see that DeepNestPort has a framework, but nothing appears documented (really sorry if I've completely missed it). Otherwise DeepNestSharp looks nice too (I like the local DXF handling), but couldn't see anything offline for it? 

Thanks again.

abebeos, 2023-10-03:
those are merely pointers on how you cound implement this:

run from cli

https://github.com/progand/Deepnest/tree/console
https://github.com/odemakov/Deepnest/commit/94e72f775b8649467025aca5f9c764ce63bb4e61

2nd link looks easy to try (note that I'm out of coding currently, lack of dev-funds here.)