deepnest-io/Deepnest #138: Segmentation fault on AppImage for Deepnest-1.0.5-linux.zip

andres-hurtado-lopez, 2024-01-22:
Hi:

I was trying to download and run from Deepnest's webpage the Appiamge from 

https://deepnest.io/Deepnest-1.0.5-linux.zip

I gave execution permissions to the appimage a single message appeared:

`andresh@andres-hurtado-lopez:~/appimages$ chmod +x Deepnest-1.0.5-x86_64.AppImage
andresh@andres-hurtado-lopez:~/appimages$ ./Deepnest-1.0.5-x86_64.AppImage
Segmentation fault`

Because of that I ran it with an strace to see what was happening inside an I got this:

`andresh@andres-hurtado-lopez:~/appimages$ strace -f ./Deepnest-1.0.5-x86_64.AppImage
execve("./Deepnest-1.0.5-x86_64.AppImage", ["./Deepnest-1.0.5-x86_64.AppImage"], 0x7fff77a4e9c8 /* 43 vars */) = 0
brk(NULL)                               = 0x1577000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f6e2ff17000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=153571, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 153571, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f6e2fef1000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=14480, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 16400, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6e2feec000
mmap(0x7f6e2feed000, 4096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f6e2feed000
mmap(0x7f6e2feee000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f6e2feee000
mmap(0x7f6e2feef000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f6e2feef000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=121280, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 123280, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6e2fecd000
mmap(0x7f6e2fed0000, 77824, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f6e2fed0000
mmap(0x7f6e2fee3000, 28672, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16000) = 0x7f6e2fee3000
mmap(0x7f6e2feea000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c000) = 0x7f6e2feea000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=14480, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 16400, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6e2fec8000
mmap(0x7f6e2fec9000, 4096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f6e2fec9000
mmap(0x7f6e2feca000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f6e2feca000
mmap(0x7f6e2fecb000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f6e2fecb000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\220x\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=1926256, ...}, AT_EMPTY_PATH) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 1974096, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6e2fce6000
mmap(0x7f6e2fd0c000, 1396736, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x26000) = 0x7f6e2fd0c000
mmap(0x7f6e2fe61000, 344064, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x17b000) = 0x7f6e2fe61000
mmap(0x7f6e2feb5000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1cf000) = 0x7f6e2feb5000
mmap(0x7f6e2febb000, 53072, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f6e2febb000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f6e2fce4000
arch_prctl(ARCH_SET_FS, 0x7f6e2fce5040) = 0
set_tid_address(0x7f6e2fce5310)         = 903037
set_robust_list(0x7f6e2fce5320, 24)     = 0
rseq(0x7f6e2fce5960, 0x20, 0, 0x53053053) = 0
mprotect(0x7f6e2feb5000, 16384, PROT_READ) = 0
mprotect(0x7f6e2fecb000, 4096, PROT_READ) = 0
mprotect(0x7f6e2feea000, 4096, PROT_READ) = 0
mprotect(0x7f6e2feef000, 4096, PROT_READ) = 0
mprotect(0x7f6e2ff49000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7f6e2fef1000, 153571)          = 0
[... 380 lines cut ...]
Codename:	trixie
andresh@andres-hurtado-lopez:~/appimages$ uname -a
Linux andres-hurtado-lopez 6.6.11-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.11-1 (2024-01-14) x86_64 GNU/Linux
andresh@andres-hurtado-lopez:~/appimages$  neofetch
       _,met$$$$$gg.          andresh@andres-hurtado-lopez
    ,g$$$$$$$$$$$$$$$P.       ----------------------------
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux trixie/sid x86_64
 ,$$P'              `$$$.     Host: ZenBook UX425QA_UM425QA 1.0
',$$P       ,ggs.     `$$b:   Kernel: 6.6.11-amd64
`d$$'     ,$P"'   .    $$$    Uptime: 3 hours, 11 mins
 $$P      d$'     ,    $$P    Packages: 3781 (dpkg)
 $$:      $$.   -    ,d$$'    Shell: bash 5.2.21
 $$;      Y$b._   _,d$P'      Resolution: 1920x1080
 Y$$.    `.`"Y$$$$P"'         WM: i3
 `$$b      "-.__              Theme: Adwaita [GTK3]
  `Y$$                        Icons: Adwaita [GTK3]
   `Y$$.                      Terminal: alacritty
     `$$b.                    CPU: AMD Ryzen 9 5900HX with Radeon Graphics (16) @ 4.680GHz
       `Y$$b.                 GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series
          `"Y$b._             Memory: 4134MiB / 15395MiB
              `"""

`

What I'm doing wrong ?

bosd, 2024-01-30:
Hi,
Thanks for your interest in deepnest. 
I'm the initiator of this community based fork of deepnest.
Since the original software is abondened by the author.

The appimage you downloaded is from the original work an author.
It is not supported here.

Sadly, this community is facing issues, and slipping down into abondonware. see #139 

But well, the source code is here is more recent. You may find your luck in building it yourself.

abebeos, 2024-06-15:
@andres-hurtado-lopez , you've downloaded an old, outdate binary, which possibly does not work in general anymore.

Try https://github.com/deepnest-io/Deepnest/releases the 1.2.0, i should work on linux.

closed 2024-06-15 by abebeos