deepnest-io/Deepnest #139: This is back abondonware

bosd, 2024-01-30:
This repo and organization is supposed to be the community based fork of deepnest.
Which has been abondoned by the original author.

**Brief history**
I've noticed that jack000 the original author of deepnest was no longer active maintaining the software and repo.
I've created this repo with the purpose of a community based fork. Ultimatly wanting to have multuple people to be able to contribute, merge and develop the community and application further.
With the ultimate goal of making the organization vonurable on a single person.
While lacking the time or technical capabilities, we've sponsored a developer to do the heavy lifting.
Cleaning of code and making it compatible with more recent builds and os's
Until we ran out of budget.

Sadly, I have to conclude we we've failed our mission on this community based fork.
Judging by his last commit the developer we hired, seems to have left github.
https://github.com/deepnest-io/Deepnest/pull/122

Whilst he took over this repo and taking away my admin rights on this repo and organization.
Due to various indifferences.
(basically, and open source software and community is based on volunteerism and collaboration, it just cannot be managed as a commercial application)

At the time I created this fork, I have chosen the unfortunate name of deepnest-io. Now we are even unable to change it into deepnest-ce. Which more clearly indicates this is the community based fork instead and not related to deepnest.io or the repo of the original author.
With a lack of active members, admin/owner rights.
We are finding this software to slip down to abandonware. :(

radevgit, 2024-01-31:
I guess, you can always fork the repo/organization and continue from there.

abebeos, 2024-06-15:
@bosd , why didn't you contact me via email?

Anyways, latest published release fixed several issues and should work on linux.

closed 2024-06-16 by abebeos