deepnest-io/Deepnest #140: [Feature request] Option to inherit initial positions from the SVG

gsuberland, 2024-02-01:
Sometimes I've got a pretty good packing of objects in a sheet, and I would like Deepnest to optimise it further.

A good example would be these pieces:

[image]

It isn't super clear from the image, but there's actually some small spacing between these pieces, and no line merging has been done. It would be helpful to be able to put this into Deepnest and have it fine-tune the positioning and do the line merging.

However, Deepnest has a ton of trouble with this shape due to starting from scratch, and it takes a large number of iterations for it to even place 60 of the 86 pieces on the board, even after carefully tweaking parameters.

It would be nice if there was a config option to inherit the initial packing solution from the imported SVG/DXF, and optimise further from there given the settings provided.

gsuberland, 2024-02-01:
Sorry, for some reason the new issue only let me open it as a bug, and auto-added the bug label.

lazaridiscom, 2024-06-18:
@gsuberland , my understanding is that this would be:

=> Provide a "Fine-Tune/Merge-Lines only" option

### Workaround

There is possibly(!) a workaround for this, by changing the nesting configuration in a way that it does not allow replacement. Not sure though how (or even if) this can be achieved.

abebeos, 2024-06-18:
@gsuberland , my understanding is that this would be:

=> Provide a "Fine-Tune/Merge-Lines only" option

### Workaround

There is possibly(!) a workaround for this, by changing the nesting configuration in a way that it does not allow replacement. Not sure though how