deepnest-io/Deepnest #142: DeepNest quantity update automatically

pupotto, 2024-06-06:
Good morning,
I downloaded your latest version 1.2.0 and in the meantime I congratulate you...
Please, I need to know if the quantity box of the individual pieces can be updated by reading from the file name.
Let me explain better, in the names of the imported dxf files I would find the indication of the quantities, it would be possible that your program at the same time as I import the dxfs could read a part of the file name which includes the quantities of pieces and automatically fill the field?
Thanks in advance
Marco

abebeos, 2024-06-15:
after fixing some things, development was stopped, mainly due to the disappearance of the initial sponsors and resulting lack of budget.

Other than that, some automation of the quantity-update should be doable, but it is currently not of any priority.

abebeos, 2024-06-15:
@pupotto , this could be done within a few days (would be in a new release 1.2.1) for a budget of $200

### Doable

`myFile.dxf` => quantity `1`

`myFile.33.dxf` => all quantities are set to `33`

### Alternative Solution Path

a (possibly more elegant) alternative would possibly be:

- provide UI functionality to be able to set the quantity for all selected items (manually select, or all by "Select All", then change the number for all). This would have the benefit of being able to select a subset of items and change their quantity at once.

bosd, 2024-07-08:
@PhilipsCodeHub 
Thanks for sharing your solution.
Can you open a PR for this?

Dexus, 2024-12-21:
You can find the fork at https://github.com/deepnest-next/deepnest

The org https://github.com/deepnest-io was overtaken by an asshole by @abebeos and @lazaridiscom his alter ego account, which deleted now anything and resetting the contributions back to a version around v1.2.0 at his moment. And also kicked in the last days all members of the Organisation. 

I will rewrite in the future the complete app and make it ready for the future.

```

  "login": "abebeos",
  "id": 165260843,
  "created_at": "2024-03-28T08:34:11Z",
  "updated_at": "2024-12-12T17:04:50Z"

and

  "login": "lazaridiscom",
  "id": 62181287,
  "created_at": "2020-03-14T15:33:46Z",
  "updated_at": "2024-07-19T11:50:07Z"

```

I would not trust the deepnest-io org in any ways in the future as long there his accounts have ownership.