deepnest-io/Deepnest #146: Tab button ruins the interface so as currently running nesting (Windows)

ClockworkBastard, 2024-07-05:
While interface is pleasant and clear, with fancy animation... It still use standard Windows' window behavior. Namely pressing [TAB] key for multiple times jumps across all (not really all) buttons and input boxes.
It totally ruins the interface, and it even would be okay if we can get back to our nesting window (part), but no. It doesn't count buttons like "Stop Nesting" as button to jump to.
[image]
[image]

abebeos, 2024-12-27:
@Dexus , 
h
i
j
a
c
k
!

Dexus, 2024-12-27:
Keine Sorge, wurde schon adressiert. In v2 bei https://www.deepnest.net wird es enthalten sein.

abebeos, 2024-12-27:
so can I safely delete? no information loss?