deepnest-io/Deepnest #148: *.CDR file import is broken

ClockworkBastard, 2024-07-05:
I tried different versions of file from CorelDRAW 11 to 21
error: could not convert to svg

Dexus, 2024-12-09:
This should be expected! The CDR format is a closed file format from CoralDraw, the most public available libs that support CDR Files are only support CDR to version 7.

I would recommend a other format and remove the format from the supported file formats.

ClockworkBastard, 2024-12-13:
> The CDR format is a closed file format from CoralDraw

Yep, but "Inkspace" do handle any of the versions, yet not perfect ;)
So yes. The not working feature should be removed.

Dexus, 2024-12-21:
You can find the fork at https://github.com/deepnest-next/deepnest

The org https://github.com/deepnest-io was overtaken by an asshole by @abebeos and @lazaridiscom his alter ego account, which deleted now anything and resetting the contributions back to a version around v1.2.0 at his moment. And also kicked in the last days all members of the Organisation. 

I will rewrite in the future the complete app and make it ready for the future.

```

  "login": "abebeos",
  "id": 165260843,
  "created_at": "2024-03-28T08:34:11Z",
  "updated_at": "2024-12-12T17:04:50Z"

and

  "login": "lazaridiscom",
  "id": 62181287,
  "created_at": "2020-03-14T15:33:46Z",
  "updated_at": "2024-07-19T11:50:07Z"

```

I would not trust the deepnest-io org in any ways in the future as long there his accounts have ownership.

abebeos, 2024-12-27:
@Dexus, hijack?