deepnest-io/Deepnest #149: add with labels in deepnest from dxf file

bidreshivaprasad, 2024-09-07:
add with labels in deepnest from dxf file

bosd, 2024-09-07:
Can you describe the bug or issue your encountering with more information?

Dexus, 2024-12-09:
> add with labels in deepnest from dxf file 

This would need to be a complete new system that could support it. But if you can provide examples what you excepted to get, it could be helpful to us to check if it is a thing we can implement.

Dexus, 2024-12-21:
You can find the fork at https://github.com/deepnest-next/deepnest

The org https://github.com/deepnest-io was overtaken by an asshole by @abebeos and @lazaridiscom his alter ego account, which deleted now anything and resetting the contributions back to a version around v1.2.0 at his moment. And also kicked in the last days all members of the Organisation. 

I will rewrite in the future the complete app and make it ready for the future.

```

  "login": "abebeos",
  "id": 165260843,
  "created_at": "2024-03-28T08:34:11Z",
  "updated_at": "2024-12-12T17:04:50Z"

and

  "login": "lazaridiscom",
  "id": 62181287,
  "created_at": "2020-03-14T15:33:46Z",
  "updated_at": "2024-07-19T11:50:07Z"

```

I would not trust the deepnest-io org in any ways in the future as long there his accounts have ownership.

abebeos, 2024-12-27:
@Dexus , you forgot this one (no backlink to your issue)

Dexus, 2024-12-27:
See no reason for this, there is no bug and no further reaction.

abebeos, 2024-12-27:
k