deepnest-io/Deepnest #150: The placements result return with the same sheetid

DayTimeDreamer, 2024-10-11:
deepnest-v1.2.0-win32-x64

Add 25 parts with 50mm*50mm and 5 sheet with 240mm*240mm, then start nesting.  All the placements result return the same "sheetid": 4.
In my opinion, the sheetid should be unique.

{
    "placements": [
        {
            "sheet": 1,
            "sheetid": 4,
            "sheetplacements": [
                {
                    "x": 0,
                    "y": 141.73228128725697,
                    "id": 1,
                    "rotation": 270,
                    "source": 0,
                    "filename": null
                },
                {
                    "x": 2.842170943040401e-14,
                    "y": 283.46456480000006,
                    "id": 2,
                    "source": 0,
                    "rotation": 270,
                    "filename": null,
                    "mergedLength": 141.7322835,
                    "mergedSegments": [
                        [
                            {
                                "x": 2.385911331785977e-15,
                                "y": 141.73228130000007
                            },
                            {
                                "x": 141.7322835,
                                "y": 141.73228130000004
                            }
                        ]
                    ]
                },
                {
                    "x": 2.842170943040401e-14,
                    "y": 425.19684830000006,
                    "id": 0,
                    "source": 0,
                    "rotation": 270,
                    "filename": null,
                    "mergedLength": 141.7322835,
                    "mergedSegments": [
                        [
                            {
                                "x": 141.7322835,
                                "y": 283.46456480000006
                            },
                            {
                                "x": 141.7322835,
                                "y": 283.46456480000006
                            }
                        ],
                        [
                            {
                                "x": 2.385911331785977e-15,
                                "y": 283.46456480000006
                            },
                            {
                                "x": 141.7322835,
                                "y": 283.46456480000006
                            }
                        ]
                    ]
[... 839 lines cut ...]
                            {
                                "x": 283.4645648,
                                "y": 425.19684989999996
                            }
                        ],
                        [
                            {
                                "x": 283.46456639999997,
                                "y": 425.19684989999996
                            },
                            {
                                "x": 283.4645648,
                                "y": 425.19684989999996
                            }
                        ]
                    ]
                }
            ]
        }
    ],
    "fitness": 928562.4051891366,
    "area": 462828.4456161795,
    "mergedLength": 4960.629920330796,
    "index": 2
}

DayTimeDreamer, 2024-10-15:
It's correct in deepnest-v1.0.5.

abebeos, 2024-12-05:
@DayTimeDreamer , for some reason, github did not ping me here.

anyways, I'm not active anymore, but if you can confirm that the bug was introduced in v1.1.5, i could take a quick look.

Dexus, 2024-12-21:
You can find the fork at github.com/deepnest-next/deepnest

The org github.com/deepnest-io was overtaken by an asshole by @abebeos and @lazaridiscom his alter ego account, which deleted now anything and resetting the contributions back to a version around v1.2.0 at his moment. And also kicked in the last days all members of the Organisation. 

I will rewrite in the future the complete app and make it ready for the future.

```

  "login": "abebeos",
  "id": 165260843,
  "created_at": "2024-03-28T08:34:11Z",
  "updated_at": "2024-12-12T17:04:50Z"

and

  "login": "lazaridiscom",
  "id": 62181287,
  "created_at": "2020-03-14T15:33:46Z",
  "updated_at": "2024-07-19T11:50:07Z"

```

I would not trust the deepnest-io org in any ways in the future as long there his accounts have ownership.