deepnest-io/Deepnest #152: Help! How to convert svg to dxf with kabeja?

DayTimeDreamer, 2024-10-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Dexus, 2024-12-09:
use inkscape or something similar.

Dexus, 2024-12-11:
Will close this for #165

closed 2024-12-11 by Dexus