deepnest-io/Deepnest #156: Canberra Gtk Module issue

devianpctek, 2024-11-17:
**Describe the bug**
I'm trying to run Deepnest on Debian Trixie, but it won't run as it gives me a segmentation fault, trying to load Canberra-gtk-module, I try installing the module libcanberra-gtk-module, but doesn't exist, it's replaced by libcanberra-gtk3-module however still getting the same error.

**To Reproduce**
Steps to reproduce the behavior:
1. Open terminal
2. Unzip package
3. go into Directory
4. give executable permission to the appimage
5. run ./appimage
6. Gtk-Message: 11:43:03.597: Failed to load module "canberra-gtk-module"
Segmentation fault

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Debian Trixie

**Additional context**
Add any other context about the problem here.

Dexus, 2024-12-09:
It needs to checked if the electron version is supporting it.

Maybe it can handled with electron 33.2+

abebeos, 2024-12-27:
@Dexus

Dexus, 2024-12-27:
ok ok @abebeos as you wish: @devianpctek sorry @abebeos can't help you here. In my fork, which you can find behind www.deepnest.net, there will be a Linux variant from version 2.0.0, which should work on all mainline Linux versions, at least I'm trying. For dependencies that come from Electron we unfortunately can't deal with them directly. But I think we will be able to provide a suitable PKG.

abebeos, 2024-12-27:
Hey @Dexus , i meant "copy the issue over to your repo". I'm shredding...

Dexus, 2024-12-27:
Lass dich nicht abhalten.

abebeos, 2024-12-27:
Breauchst' den Text nich'?