deepnest-io/Deepnest #157: Importing Not Working

Eric142-ai, 2024-11-25:
Hello,

I've been using Deepnest for the past week to nest and it has been working as intended. This morning, I proceeded to import files as normal but none of the shapes have been registered by the import (they do not appear). I've tried previous files that have worked in the past and those haven't imported either.
Double checked if they were simply invisible by adding a sheet to nest on - no shapes, just the sheet.
Restarted PC, checked latest update version, didn't work.

Tl;dr There seems to be a problem with the importing, at least on my end.

Expected:
- Files to import and for the shapes to appear

**Screenshots**
[image]
[image]
[image]

^Checked if shapes were present by nesting on an added sheet

Desktop: 
- Windows 11 Pro
- Version: 23H2
- Browser: Microsoft Edge
- Deepnest: Both v1.0.5 (what I was using before) and v1.2.0 (what I tried after looking at latest update), both don't work.

abebeos, 2024-12-05:
> maybe that went offline.

There was a paywall added, documented within https://deepnest.io/, but seems it is removed now.

@Eric142-ai , @fluxor99232 , is this still happening?

Eric142-ai, 2024-12-06:
Hi both,

Thank you for the info/input. The dxf importing is working again so the issue seems to be resolved.

closed 2024-12-06 by Eric142-ai