deepnest-io/Deepnest #163: [ELECTRON] Consider dropping electron support

abebeos, 2024-12-09:
work form @ShaMan123 converted deepnest to a node app (http server).

Electron support was dropped:

https://github.com/ShaMan123/Deepnest/pull/2

Consider adopting those changes, especially if shaman plans to further maintain/evolve this version.

Dexus, 2024-12-09:
I'm currently on electron 33.2 and porting some of the functions to a rust libary for more performance. NodeJS is still to slow in my opinion.

abebeos, 2024-12-09:
No sure what you mean here. "you are on electron 33.2" - you mean with Deepnest?

Dexus, 2024-12-09:
I have updated DeepNest to electron 33.2 and ported some functions to a rust lib that export a nodejs binding. At least I'm trying to port the code to TS and port it more and more with the CPU hungry tasks to rust.

ShaMan123, 2024-12-09:
I need a function that I can call, not UI. That is why I ported the code to nodejs. 
It is quick enough for me since I nest elements<50.

Dexus, 2024-12-09:
I'm nesting parts around 300+ and its took to much time. also there are other libs, that do things faster, than the current JS. 
jagua-rs is also a nice nesting lib in rust, but has not yet the complete features to so what we need here.

lazaridiscom, 2024-12-09:
The problem is that over the years, many people worked on custom versions of deepnest (see https://github.com/orgs/deepnest-io/people).

At one point I got some budget to integrate results, and move towards newer electron versions. Stopped at around 13. Updated yesterday (quite untested) to electron 19 (from @ShaMan123 ).

@Dexus , could you point me to your repo? Is it a fork of this repo here? I'm curious if your electron 33 efforts could be integrated here into this repo.

@ShaMan123 's work is very nice, but integrating into the main repo needs some effort. Me not sure if dropping electron fully is a good thing.

abebeos, 2024-12-09:
The problem is that over the years, many people worked on custom versions of deepnest (see https://github.com/orgs/deepnest-io/people).

At one point I got some budget to integrate results, and move towards newer electron versions. Stopped at around 13. Updated yesterday (quite untested) to electron 19 (from @ShaMan123 ).

@Dexus , could you point me to your repo? Is it a fork of this repo here? I'm curious if your electron 33 efforts could be integrated here into this repo.

@ShaMan123 's work is very nice, but integrating into the main repo needs some effort. Me not sure if dropping electron fully is a good thing.

abebeos, 2024-12-09:
> am too tired of what I put into it and have no value in it so I stopped there.

very well. note that I do not work on code-level here anymore. I made an exception to integrate the electron 19 work.

possibly you can revise your work in future, thus it becomes mergable.

Dexus, 2024-12-09:
Gladly, but I have to clean up my "rubbish" first so that only the parts that I needed for Electron 33.2 remain. At the moment everything is still on my internal GIT server but I can probably push it to GitHub here tomorrow evening.

abebeos, 2024-12-09:
> I can probably push it to GitHub here tomorrow evening.

very nice. don't invest too much time with cleanup, you can do it later anyway.

I'll will invest some time then to see if the changes can be integrated easily.

Dexus, 2024-12-09:
That's not a big fish I have to move, I just have "too few" commits and more after the weekend commits xD

Dexus, 2024-12-10:
@abebeos with your latest update there was not much to change for me. but have added the patch-package dev-deps and a patch for the electron-settings if you think we will undo the remove of it. my PR #166 & #167

ShaMan123, 2024-12-10:
BTW my fork requires node>=20 due to [CustomEvent](https://github.com/nodejs/node/pull/43885)

abebeos, 2024-12-10:
> node>=20

that's fine, support for ancient versions can be dropped anyways at some point

abebeos, 2024-12-10:
will look later tonight at the PR's, hope we can bring this thing finally up-to-date and more usable!

abebeos, 2024-12-10:
Looks like electron can stay, closing, see #166

closed 2024-12-10 by abebeos