deepnest-io/Deepnest #165: [request] offline dxf to svg (and svg to dxf)

Dexus, 2024-12-09:
Only a reminder for me.

Support (light) offline support:

- Add dxf2svg
- maybe add svg2dxf if no big changes needed.

abebeos, 2024-12-10:
@Dexus , I sent you and invite.

Dexus, 2024-12-21:
You can find the fork at github.com/deepnest-next/deepnest

The org github.com/deepnest-io was overtaken by an asshole by @abebeos and @lazaridiscom his alter ego account, which deleted now anything and resetting the contributions back to a version around v1.2.0 at his moment. And also kicked in the last days all members of the Organisation. 

I will rewrite in the future the complete app and make it ready for the future.

```

  "login": "abebeos",
  "id": 165260843,
  "created_at": "2024-03-28T08:34:11Z",
  "updated_at": "2024-12-12T17:04:50Z"

and

  "login": "lazaridiscom",
  "id": 62181287,
  "created_at": "2020-03-14T15:33:46Z",
  "updated_at": "2024-07-19T11:50:07Z"

```

I would not trust the deepnest-io org in any ways in the future as long there his accounts have ownership.