deepnest-io/Deepnest #175: ERROR: could not find a version of Visual Studio 2017 or newer to use

abebeos, 2024-12-13:
I cannot rebuild on my machince, most possibly a newer version of the visual studio build tools are needed

```
gyp verb find VS unsupported version "15.9.33529.398" found at "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp verb find VS could not find a version of Visual Studio 2017 or newer to use
gyp verb find VS not looking for VS2017 as it is only supported up to Node.js 21
gyp verb find VS not looking for VS2017 as it is only supported up to Node.js 21
gyp verb find VS not looking for VS2017 as it is only supported up to Node.js 21
gyp verb find VS not looking for VS2015 as it is only supported up to Node.js 18
gyp verb find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS
gyp ERR! find VS msvs_version was set from command line or npm config
gyp ERR! find VS - looking for Visual Studio version 2017
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS could not use PowerShell to find Visual Studio 2017 or newer, try re-running with '--loglevel silly' for more details.
gyp ERR! find VS
gyp ERR! find VS Failure details: undefined
gyp ERR! find VS unsupported version "15.9.33529.398" found at "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
gyp ERR! find VS not looking for VS2017 as it is only supported up to Node.js 21
gyp ERR! find VS not looking for VS2017 as it is only supported up to Node.js 21
gyp ERR! find VS not looking for VS2017 as it is only supported up to Node.js 21
gyp ERR! find VS not looking for VS2015 as it is only supported up to Node.js 18
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS
gyp ERR! find VS valid versions for msvs_version:
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
Error: Could not find any Visual Studio installation to use
    at VisualStudioFinder.fail (P:\work\Deepnest\node_modules\@electron\node-gyp\lib\find-visualstudio.js:118:11)
    at VisualStudioFinder.findVisualStudio (P:\work\Deepnest\node_modules\@electron\node-gyp\lib\find-visualstudio.js:74:17)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async createBuildDir (P:\work\Deepnest\node_modules\@electron\node-gyp\lib\configure.js:112:18)
    at async process.<anonymous> (P:\work\Deepnest\node_modules\@electron\rebuild\lib\module-type\node-gyp\worker.js:27:13)

× Rebuild Failed

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild 'P:\work\Deepnest'

Error: node-gyp failed to rebuild 'P:\work\Deepnest'
    at ChildProcess.<anonymous> (P:\work\Deepnest\node_modules\@electron\rebuild\lib\module-type\node-gyp\node-gyp.js:121:24)
    at ChildProcess.emit (node:events:518:28)
    at ChildProcess._handle.onexit (node:internal/child_process:294:12)

```

Dexus, 2024-12-13:
You need vs2022 c++ for desktop at least that is currently my env

Dexus, 2024-12-13:
I have a full setup I think. Its a VM on my Server.

abebeos, 2024-12-13:
### Solution

install vs2022, e.g. via minimukm install here: https://visualstudio.microsoft.com/downloads/#build-tools-for-visual-studio-2022

closed 2024-12-13 by abebeos