deepnest-io/Deepnest #185: Notable Fork:

abebeos, 2024-12-17:
https://github.com/deepnest-next/deepnest

Dexus, 2024-12-17:
Hello abebeos or lazaridiscom, 

could this whole power struggle and hostile takeover of this organisation be reversed?

Now that you've banned everyone except for your own accounts and me, we should talk about how we can de-escalate the whole thing again. And put the organisation back in the right hands.

I know that you were previously hired as a freelancer for this project, but you didn't start it and now yesterday you finally took one step too many in the absolutely wrong direction.

Therefore I ask you to release the organisation again and return it to bosd or me so that I can take the appropriate steps. When I see that you are already linking to the new org, it looks to me at least as if you are sorry that you overreacted.

With this in mind, I would be grateful, and I'm not just speaking for myself, if you would be so kind as to reverse your mistakes immediately, as far as possible.

Thank you very much,
Josef Fröhle

abebeos, 2024-12-17:
you derive faulty conclusions based on incomplete/faulty information.

I'm not in the mood to correct your analytical nonsense.

The main fact is: I revived Deepnest in a 99% solo-run (see history), payment stopped, I continued unpaid in a solo-run up to some point.

So, I'll not hand over any work for which I was not payed - that simple.

But seeing 2 new devs, my plan was to hand over full control to you 2 devs, after(!) seeing that you 2 have the discipline to follow standard-procedures (mainly to produce 'chirurgical' PRs/commits. You don't have neither the discipline, nor 
 the will to learn it.

So, I placed the link to you project in the hope that the two uncoordinated/undisciplined 'stallions' will still get a decent result, thus (most importantly) users can choose which fork to use.

I'll release this org once I get the payment (don't remember, something in the region of $3K to $5K). Otherwise I shut it slowly down to finally get my closure here.

abebeos, 2024-12-20:
@Dexus, you may convince @bosd to arrange $2K. For this i would transfer ownership to  whoever wants to go on. This way the stars/forks/watches/links are not lost.

I estimate that within the next 7 days I'll shred all this (personal need, so I can move on with my personal projects within winter-season).

Dexus, 2024-12-21:
Do what you want... At least you won't make any friends. But you probably don't have any, otherwise you would realise that.

abebeos, 2024-12-21:
Not sure why it's so difficult for you to understand "commercial context". Anyways.

Dexus, 2024-12-21:
Commercial context? I've been running several companies for 15 years, so I know when I'm still in charge and when I'm not. If I continue to do so, it's my loss and not the customer's. So I don't understand your approach at all. 

If it's about the money you think you're entitled to, file a lawsuit and don't just take over the organisation. What you're doing is and remains blackmail in this context. 

If you weren't using blackmail, people could talk to you, but this way you're closing off all possibilities.

abebeos, 2024-12-21:
"think you're entitled to"? Lawsuit? Blackmail? Always fascinating to see how logic can go nirvana...

My reply to you: just piss off, you little shit.

Note to other readers: 

've identified some other folks that looks able to take the org over. So possibly I'll avoid shredding.