deepnest-io/Deepnest #189: Consider restoring electron-settings

abebeos, 2024-12-23:
Replaced with api compatible on-the-fly code within:

* https://github.com/deepnest-io/Deepnest/pull/188

A possible patch to keep `electron-settings` is mentioned here:

* https://github.com/deepnest-io/Deepnest/pull/166#discussion_r1878004652

abebeos, 2024-12-25:
@Dexus , what was this `electron-settings.patch` about, and where did you get it? didn't find anything quickly.

Dexus, 2024-12-25:
Need patch-package installed with script install.
Then the patch placed into patch directory. It's injecting the @electron/remote package because its not in the v4 and v5 never released which should fix it.

Send from 🛋️

abebeos, 2024-12-25:
did you create the patch yourself? I cannot find it, and it's gone from the PR.

Dexus, 2024-12-25:
yepp, I created it my self.

abebeos, 2024-12-25:
electron-settings should work, without patching. Seems there's no documentation about any incompatibility.

Dexus, 2024-12-25:
Since the `const {remote} = require('electron');` does not export `remote` anymore, you need `@electron/remote` package to have the same features. I did also not found a direct changelog that this was removes as breaking change but it was done somehow and somewhere...

Dexus, 2024-12-25:
If you want to cheey-pick: SHA:16926e6b0f205f1eca0a7e1bccab24bcb828c1ac

abebeos, 2024-12-25:
this sha seems to be the wrong one (https://github.com/deepnest-io/Deepnest/pull/167/commits/16926e6b0f205f1eca0a7e1bccab24bcb828c1ac) but does not matter anyways.

I've updated the issue text with a conclusion/consideration, looks like we should drop `electron-settings`, what do you think?

Dexus, 2024-12-25:
https://github.com/deepnest-io/Deepnest/blob/16926e6b0f205f1eca0a7e1bccab24bcb828c1ac/patches/electron-settings.patch

abebeos, 2024-12-25:
getting messy, cannot get the patch to work. Closing for now as `wontfix`, as electron-settings is quasi-unmaintained.

closed 2024-12-25 by abebeos

Dexus, 2024-12-25:
Everything that lies outside your comfort zone is too much for you?  Oha

abebeos, 2024-12-25:
Josef, hammse Dir ins Gehirn geschissen order watt?

Please have more precision if you feel the need to off-topic babble.

tried your patch out of curiosity. Main point is

>  electron-settings is quasi-unmaintained.

As for confort-zone:

Yes, diffs like this are far out of my comfort-zone:

https://github.com/deepnest-io/Deepnest/compare/16926e6b0f205f1eca0a7e1bccab24bcb828c1ac..b7f5c2b6be1e340df53468e336a962d8a255bba2

comfort-zone: cherry-pickable commit.

Dexus, 2024-12-25:
It was meant as a joke. But well, you couldn't read it that way. 

I'll just withdraw so that you don't have to read any off-topic drivel.

abebeos, 2024-12-25:
> It was meant as a joke.

My reply as well (combined with real stuff)