deepnest-io/Deepnest #199: Lost in precision when in metric units

Whateverman2012, 2024-12-26:
Hi,

  I mainly use mm but I noticed a slight difference between results requested in inches and in mm.
  Imperial units seem to create the desired output.
  However, metric units output is a bit off.
  So I tried to sneak a bit into the code and make the scale value a float instead of an integer.
  Now both results are the same.
  I guess it has something with a rounding effect of that integer which distorts values.
  I didn't look any further.  

  Here what I tried:

  In Deepnest/main/deepnest.js line 522:

			if(c.scale && parseInt(c.scale) > 0){
				config.scale = parseInt(c.scale);
			}
  ->
			if(c.scale && parseFloat(c.scale) > 0){
				config.scale = parseFloat(c.scale);
			}

  Regards

Dexus, 2024-12-26:
Good Catch!

abebeos, 2024-12-27:
@Dexus , "hijacking" issues - that's a new one!

Shouldn't you fix "in the repo of the issue occurrence", and then fetch the change to your fork?

Dexus, 2024-12-27:
No, I don't have to. I don't see this as an active main fork at the moment either. You delete too much and do hot-headed things that are absolute nonsense. And since you've taken over the organisation in a really cheeky way, I don't see it as my duty to actively contribute anything here. As long as this is not reversed, I will not open the first PR here. You're just lucky to still be an owner because GitHub won't intervene directly without a court order, but that will happen in January. Everything is in progress.

I tried to start it again, but then issues etc. were deleted again. I don't want to do that any more.

abebeos, 2024-12-27:
> You're just lucky to still be an owner because GitHub won't intervene directly without a court order, but that will happen in January. Everything is in progress.

Oh my, what stuff did fry your brain? Or are you a natural?

Dexus, 2024-12-27:
> > You're just lucky to still be an owner because GitHub won't intervene directly without a court order, but that will happen in January. Everything is in progress.
> 
> Oh my, what stuff did fry your brain? Or are you a natural?

That's exactly what I mean, a Linus can talk like this to his contributors, but not you little light.

abebeos, 2024-12-27:
> Linus

Torwalds? Lacks precision, lacks analytical strength. lacks "intuitive design", lacks ability to delegate, lacks, lacks, lacks. Linux development processes suck even more that git.

Give some better example.

> to his contributors

I see you as an (nearly) equal. In fact, the combination of the two of use result in "higher grade performance". 

see "Brute-Force fast workers and highly disciplined structural/strategic works complement and need(!) each other" from https://github.com/deepnest-io/Deepnest/blob/master/CONTRIBUTING.md

But... we tend to need far too much time to give in. And I must switch to some commercial tasks after too much of a (burn-out,/midlife-crisis break). So, speeding everything up here.