deepnest-io/Deepnest #201: Dexus: could you repeat PR 18?

abebeos, 2024-12-27:
The diff of `deepnets.js` is overloaded with formatting changes, hiding this way the relevant change:

- https://github.com/deepnest-next/deepnest/pull/18
- https://github.com/deepnest-next/deepnest/pull/18/files#diff-86ca7539e69ca9daf89253145916140053de9f76e8e72fe217639053da1a0c16

So as-is, one cannot `cherry-pick` the commit, as it would obfuscate the change.

abebeos, 2024-12-27:
(@Dexus , i cannot comment onf deepnest-next, thus filing issue here)

Dexus, 2024-12-27:
deepnest: https://github.com/deepnest-next/deepnest/pull/18/files#diff-86ca7539e69ca9daf89253145916140053de9f76e8e72fe217639053da1a0c16R830-R839 are the relevant stuff. 

If you want that I commit here, make sure you have a editor.config and all stuff is formated that way. Apart from that, what I wrote in #199 applies.

abebeos, 2024-12-27:
> are the relevant stuff.

doesn't matter. The commit needs to show only relevant stuff. To use your way of expressing things:

Only an absolute amateur or an idiot would make a tiny changes within of masses of whitespace noise.

Fix it, it's still easy to do (not many new commits)